TIS2024-FMFI/serial-transformer #4: Choose microcomputer for project

AndrejNagy, 2024-10-14:


pavelpetrovic, 2024-10-29:
Priklad komunikacie so seriovym portom z vlastnej aplikacie na Windows (c++):
  http://dai.fmph.uniba.sk/courses/pphw/new/tatrabot_studio_2.zip

Prepojenie pinov Teensy pre 2. seriovy port:

kabel:

cierny - GND
cerveny - VCC
oranzovy - TX
zlty - RX

teensy:

RX2 - 7  - oranzovy
TX2 - 8  - zlty
GND - GND - cierny

pavelpetrovic, 2024-10-29:
```
void setup() {
  
   Serial.begin(115200);
   Serial2.begin(115200);
}

int x = 0;
void loop() {
  
  Serial.print(x++);
  Serial.print(" ");
  Serial.println("Teensy is happy!");
  delay(500);
  if (Serial2.available())
  {
    char c = Serial2.read();
    Serial.print("Char: ");
    Serial.println(c);
  }
  
}
```

pavelpetrovic, 2024-10-29:
ovladace pre kabel:
  https://ftdichip.com/drivers/vcp-drivers/