TIS2024-FMFI/serial-transformer #12: návrh používateľského rozhrania

AndrejNagy, 2024-10-28:


DjMaestr0, 2024-10-28:
[image]

verzia GUI pre transformacny modul pre teleskop

DjMaestr0, 2024-11-04:
[image]
Upravena verzia

pavelpetrovic, 2024-11-05:
- ukazat logy smeruje na stranku so zoznamom suborov, ktore si pouzivatel moze downloadnut 
   (staci tuto vetu pridat pod obrazok do dokumentu Navrh)
- uvazujeme nad:  tlacidlo "downloadnuť všetky logy", "zmazať všetky logy" (ktore ale ma vyskakovaci potvrdzovacu stranku)
- zvazit ci sem nepridat aj konfiguraciu podla 3.2.5
- pridat dropdown, kde sa da vybrat casovy interval priemerovania tych dvoch zobrazenych rychlosti a dva textove labely, ktore ukazuju momentalnu hodnotu (posledny bod v grafe) vyjadrene ciselne (8.2.1)
- k editboxu Rychlost pridat tlacidlo "nastav" a este jeden read-only label zobrazujuci aktualnu platnu hodnotu (obe aj rucne nastavenu aj automaticku - ak sa momentalne vypocitava)
- pridat okienko pre univerzalny modul - staci jednoduche s tlacidlom "nacitat pravidla" a jednoducha statistika kolko paketov preslo a kolko z nich bolo modifikovanych

DjMaestr0, 2024-11-11:
[image]
[image]

pavelpetrovic, 2024-11-12:
zabudli sme na deaktivaciu prekladania - s tym vsetko sa preposiela, ale zastavit pakety netreba, takze to moze plnit ten ucel, ale malo by to byt aj v univerzalnom rezime a tiez deaktivaciu logovania - ci sa prechadzajuca komunikacia loguje alebo nie.

destruktivne akcie - ako napr. zmazanie logov - vzdy musia mat potvrdzovaci dialog.

DjMaestr0, 2024-11-18:
[image]
[image]
[image]

DjMaestr0, 2024-12-09:
[image]
[image]
aktualizovaná verzia modulu pre teleskop po upresnení zadávateľov

closed 2024-12-09 by IvanaBekesova