TIS2024-FMFI/serial-transformer #15: návrh implementácie (uml diagramy)

AndrejNagy, 2024-10-28:


AndrejNagy, 2024-11-04:
[image]

AndrejNagy, 2024-11-11:
New version:[image]

pavelpetrovic, 2024-11-12:
- prosim pridat vypoctovy modul, ktory dokaze vypocitat momentalne natocenie teleskopu podla prechadzajucich paketov, bude zaradeny aj na vstupe aj na vystupe;
- oddelit tri komponenty, ktore su mimo nas system hore, a zabalit nase dva nejakym boxom "system/subsystem"  a nazvat to "navrhovany system" alebo nieco podobne
- comm  module je vlastne konfiguracny modul a komunikuje iba s prehliadacom na PC - predpoklad je, ze to je nieco beziace vo webovom servri
- je do neho (konfiguracneho modulu) pripojene rovno aj spojenie od kameroveho softveru, ktory nepotrebuje v nasom systeme samostatny modul
- vstupny a vystupny kanal su pripojene (oba na oba) iba rovno na transformacne moduly (ako pripojit vypoctovy modul sa da nakreslit rozne)

Nemeth03, 2024-11-18:
State Diagram.

[image]

AndrejNagy, 2024-11-25:
[image]

closed 2025-02-16 by IvanaBekesova